RPi-Distro/repo #387: pixflat-icons (0.26) - known incorrect sRGB profile

scruss, 2025-02-01:
**Severity:** Low 

## Issue description

`libpng warning: iCCP: known incorrect sRGB profile` warnings are generated unexpectedly when, for example, previewing a JPEG image in OpenCV in Python.. These warnings are caused by incorrect colour profile information in some of the PiXflat icon files.

Roughly half of the PNG icon files/links in the package are affected. `pngcrush` can be used to list them:

    find /usr/share/icons/PiXflat -type f -name '*.png' -print0 | xargs -0 pngcrush -n -q 2>&1 | awk '/\/usr\/share/ {sub(/:$/,"");file=$1} /pngcrush: iCCP:/ {print file}'

The affected files include:
```
/usr/share/icons/PiXflat/24x24/actions/go-down.png
/usr/share/icons/PiXflat/24x24/actions/go-next.png
/usr/share/icons/PiXflat/24x24/actions/go-previous.png
/usr/share/icons/PiXflat/24x24/actions/go-up.png
/usr/share/icons/PiXflat/24x24/mimetypes/document.png
```

Files that do not have the issue include:
```
/usr/share/icons/PiXflat/128x128/places/folder-remote.png
/usr/share/icons/PiXflat/128x128/places/user-bookmarks.png
/usr/share/icons/PiXflat/128x128/places/user-trash-full.png
/usr/share/icons/PiXflat/128x128/places/user-trash.png
/usr/share/icons/PiXflat/160x160/places/folder-remote.png
```

Unaffected files don't appear to have embedded ICC profiles, while affected ones appear to have colour profiles added by Adobe Photoshop CC 2019 (Macintosh).

## Possible remedy

It may be sufficient to remove the ICC profile from the files. This can be done using:

    pngcrush -rem iCCP foo.png result.png

(source: [svgeesus/PNG-ICC-tests: Tests for ICC color profiles embedded in PNG images](https://github.com/svgeesus/PNG-ICC-tests?tab=readme-ov-file#to-remove-an-icc-profile))

## User reference

A user reported this problem in the thread [warning message when reading .jpg file on RPi4b with cv2 - Raspberry Pi Forums](https://forums.raspberrypi.com/viewtopic.php?t=378199)

RickyW27, 2025-02-16:
How can I apply the code to jpg file?

scruss, 2025-02-23:
> How can I apply the code to jpg file?

This has nothing to do with JPEG files