RPi-Distro/repo #388: apt fails when running on GitHub Actions

probonopd, 2025-02-08:
apt fails on `2024-11-19-raspios-bookworm-armhf-lite.img.xz` when running on GitHub Actions.

Here is my build.yml:
https://github.com/probonopd/build-mac-emu/blob/main/.github/workflows/build.yml

As you can see in [line 22](https://github.com/probonopd/build-mac-emu/actions/runs/13218686837/job/36901058697#step:6:22) of the build log, I am doing `sudo apt-get -y update`

As you can see in [line 245](https://github.com/probonopd/build-mac-emu/actions/runs/13218686837/job/36901058697#step:6:246), apt wants to download [http://raspbian.raspberrypi.com/raspbian/pool/main/g/git/git-man2.39.5-0%2bdeb12u1_all.deb](http://raspbian.raspberrypi.com/raspbian/pool/main/g/git/git-man_2.39.5-0%2bdeb12u1_all.deb)

But actually the version on the server is [http://raspbian.raspberrypi.com/raspbian/pool/main/g/git/git-man2.39.5-0%2bdeb12u**2**_all.deb](http://raspbian.raspberrypi.com/raspbian/pool/main/g/git/git-man_2.39.5-0%2bdeb12u2_all.deb)

Curiously the same does not happen when running on actual Raspberry Pi hardware.

What gives?

XECDesign, 2025-02-10:
This doesn't look like an issue with the archive.raspberrypi.com repo. I don't know enough about the hardware or environment of the runners to comment.

If it's an aarch64-only system and requires qemu-user to run armhf binaries, then that may introduce all kinds of issue.

probonopd, 2025-02-10:
Should have used arm64 from the beginning. Thanks for the spot-on hint.

closed 2025-02-10 by probonopd