RPi-Distro/repo #390: firmware-brcm80211 rpt3 seems to cause periodic network disconnects on multiple Raspberry Pi 4 (bookworm)

mathieu5500, 2025-04-25:
**Describe the bug**
After updating to `firmware-brcm80211 1:20240709-2~bpo12+1+rpt3`, both of my Raspberry Pi 4 devices began experiencing periodic disconnections from the LAN. This affects both SSH and access to a locally hosted web interface. They loose access to the server for too long. According to previous test, the wifi connection always is good, but the bitrate drops when the disconnection occur.

The devices become unreachable for a short period (timeout), then reconnect can be reach again manually. This is repeatable and happens with different SSH clients (PuTTY, Windows cmd, iPhone app).

The issue only occurs on the Pi 4 devices that received the `rpt3` update. A Raspberry Pi 5 on the same network — running the same kernel but with the previous `firmware-brcm80211 rpt2` version (manually held back) — does not disconnect at all.

This suggests a regression in the `rpt3` firmware.

---

**To reproduce**
1. On Raspberry Pi 4, upgrade to Raspberry Pi OS Bookworm with `firmware-brcm80211 1:20240709-2~bpo12+1+rpt3`.
2. Connect the device via Wi-Fi (LAN, not public internet).
3. Leave idle or attempt continuous SSH or HTTP access.
4. After some time, connection drops (timeout), and Pi becomes temporarily unreachable.

However, the issue might be triggered by something on my home network. I have a lot of multicast/broadcast spam from smart devices.

---

**Expected behaviour**
The Raspberry Pi 4 should remain stably connected to the LAN like it did with earlier firmware versions.

---

**Actual behaviour**
All affected Raspberry Pi 4 devices experience random disconnects from LAN — affecting SSH and web UI. Devices recover automatically but remain unstable and need manual reconnection (for ssh). The Raspberry Pi 5 with `rpt2` firmware stays connected without issues.

---

**System**

**Affected Raspberry Pi 4 devices (x2):**
- OS: Raspberry Pi OS Bookworm 64-bit
- Kernel: `Linux 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19)`
- Package: `firmware-brcm80211 1:20240709-2~bpo12+1+rpt3`
- Network: Wi-Fi (same LAN as other devices)

**Working Raspberry Pi 5 device:**
- OS: Raspberry Pi OS Bookworm 64-bit
- Kernel: `Linux 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1`
- VC firmware: `2025/02/12 10:51:52 — version f788aab6`
- Package: `firmware-brcm80211 1:20240709-2~bpo12+1+rpt2` (manually held)
- Stable connectivity

---

**Additional context**
- All devices are on the same LAN, surrounded by smart devices that may generate multicast or broadcast traffic.
- Attempts to isolate or mitigate the issue (disabling multicast, blocking traffic via UFW, disabling IPv6, stripping services, disabling ufw) were unsuccessful.
- The two Raspberry Pi 4s do not share identical configurations/softwares, yet both show the same issue after the `rpt3` update.

---

**Questions**
- Is this a known regression in `firmware-brcm80211 rpt3`?
- Is there an official way to downgrade or pin to `rpt2` until resolved?
- Would logs or debug output be useful for investigating further?

Thanks for your help!

XECDesign, 2025-04-25:
Does commenting out the line in `/lib/modprobe.d/rpi-brcmfmac.conf` and rebooting fix the problem?

mathieu5500, 2025-04-26:
No, it is not working. From what I understand, this line is related to wifi power save? I forgot to mention, I also already tried turning off wifi power save with a service at boot.

To test your method, I commented the line you specified on only one of the Raspberry Pi. Rebooted both Raspberry Pi and tested if they still both disconnect at the same time. They did.

Thanks for your reply!

XECDesign, 2025-04-26:
That was the only change between rpt2 and rpt3.

mathieu5500, 2025-04-28:
Alright, I will look further into it. Thanks.

mathieu5500, 2025-05-15:
I’ve identified the root cause of the issue.
During testing, one of the Raspberry Pis crashed while the other remained stable—an unusual outcome since they typically crash simultaneously. Upon investigation, I found that the crashing device was connected to the 5 GHz band, while the stable one was on 2.4 GHz. Both were frequently switching between bands on my dual-band SSID.
To further isolate the problem, I flashed a microSD card with a backup image from January 2025—prior to the brcm80211update or any other significant system changes. With this older image, the Raspberry Pi consistently connects to the 2.4 GHz band and remains stable without crashes.
At the time of testing, one device was located approximately 30 cm from the router, and the other about 6 m away. Given that the closer device still exhibited issues on 5 GHz, it suggests a problem specific to 5 GHz handling rather than weak signal strength or distance.
When I forced the Pi to connect to the 5 GHz band using router settings—even with the January image—the issue reappeared. This indicates the problem lies in 5 GHz support itself. It also appears that a change introduced between January and April 2025 enabled the Pi to start using the 5 GHz band, whereas previously it defaulted to 2.4 GHz.
In summary, the instability seems related to how the current version of the Raspberry Pi OS handles 5 GHz Wi-Fi, possibly due to changes in the wireless driver or network management behavior introduced after January 2025.