RPi-Distro/repo #391: Backport of gstreamer V4L2 seeking fix

ardera, 2025-05-08:
There was a race condition in the gstreamer video4linux2 decoder plugin that resulted in seek operations leading to some buffer management issues, causing crashes like this later on:

```
0:01:04.181753782 30180   0x7f40022700 ERROR          v4l2allocator gstv4l2allocator.c:1328:gst_v4l2_allocator_dqbuf:<v4l2h264dec3:pool1:src:allocator> buffer 1 was not queued, this indicate a driver bug.
```
```
0:03:14.468175136  9634   0x7f6401b700 ERROR         v4l2bufferpool gstv4l2bufferpool.c:1491:gst_v4l2_buffer_pool_dqbuf:<v4l2h264dec0:pool1:src> No free buffer found in the pool at index 2.
```

It is actually not that hard to trigger, you just have to do some seek operations while video playback is running. One occasion where you will definitely run into it is when doing looping video playback. E.g. when looping this video: https://flutter.github.io/assets-for-api-docs/assets/videos/bee.mp4 using [flutter-pi](https://github.com/ardera/flutter-pi) & Flutter, you will definitely see a crash after max. 5 minutes of looping.

It has been fixed in gstreamer 1.23: https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/4465
and the fix was backported to 1.22, and is present in 1.22.8 to 1.22.12: https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/5695

It would be very good if the distro package could be updated to at least gstreamer 1.22.8. I don't know who else is affected by this, but there's a lot of people complaining about this in the flutter-pi issues; and there's not really a good workaround, except disabling hardware decoding completely for affected gstreamer versions

XECDesign, 2025-05-12:
Many thanks for the good bug report. While I'm not keen on updating ahead of Debian, since it can introduce new issue, I'm happy to pull in the fix.

@6by9, any objections to adding https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/5695?

XECDesign, 2025-05-13:
I've added a patched version to the 'beta' component of our repo. If you can confirm that it resolves the issue, I can move it across to 'main'. You can enable it by adding 'beta' after 'main' in `/etc/apt/sources.list.d/raspi.list`

6by9, 2025-05-13:
Sorry for the delay - looks fine to me.

ardera, 2025-05-13:
> I've added a patched version to the 'beta' component of our repo. If you can confirm that it resolves the issue, I can move it across to 'main'. You can enable it by adding 'beta' after 'main' in `/etc/apt/sources.list.d/raspi.list`

Thanks! I just tested and I can confirm that the `gst-plugins-good` from `beta` fixes it.

XECDesign, 2025-05-13:
Thanks for testing. Shipped