RPi-Distro/repo #394: German “ä” key not working on multiple keyboards under Debian 12 (Bookworm) aarch64 on Raspberry Pi 5

ab-in, 2025-06-18:
On my Raspberry Pi 5 (Model B Rev 1.0) running Debian GNU/Linux 12 (Bookworm) aarch64, the lowercase “ä” key does not work on any connected keyboard (tested with several Bluetooth and USB keyboards, all with German layout). The issue persists regardless of keyboard model or connection type.

Details:

Device: Raspberry Pi 5 Model B Rev 1.0

OS: Debian GNU/Linux 12 (Bookworm) aarch64

Kernel: 6.12.25+rpt-rpi-2712

Keyboards tested: Multiple (Bluetooth and USB, German layout, including 65% and standard size)

Issue: Pressing the “ä” key produces no input. In xev, no KeyPress/KeyRelease event is generated for this key, only a KeymapNotify event. All other keys work as expected.

What I tried:

Verified keyboard layout (de, de-t3, de-nodeadkeys, etc.)

Used setxkbmap and reconfigured /etc/default/keyboard

Upgraded all packages (sudo apt full-upgrade)

Edited /etc/X11/xorg.conf.d/10-keyboard.conf

Set HID2HCI_ENABLED=1 in /etc/default/bluetooth

Tested with different user accounts and fresh installs

Tested on Ubuntu for Raspberry Pi: The “ä” key works perfectly there!

Expected behavior:
The “ä” key should produce the character “ä” (adiaeresis) as on other Linux distributions.

Actual behavior:
No character is produced, and no KeyPress/KeyRelease event is detected in xev.

Steps to reproduce:

Install Debian 12 (Bookworm) aarch64 on Raspberry Pi 5

Connect a German-layout keyboard (Bluetooth or USB)

Set keyboard layout to “de”

Try to type “ä” (lowercase)

Observe that nothing happens and xev shows no key event

Additional information:
This issue affects all tested keyboards and persists after all configuration and upgrade attempts. It does not occur on Ubuntu for Raspberry Pi with the same hardware.

lurch, 2025-06-18:
Do you get the same behaviour in Raspberry Pi OS and also Raspberry Pi OS Lite? (which might help narrow down whether it's a kernel problem or a Wayland problem)

ab-in, 2025-06-18:
I don't know , I have no raspiOS here and it dosn't matter because I need Debian

closed 2025-06-19 by XECDesign