RPi-Distro/repo #395: [Feature Request] Enable KVM support in Raspberry Pi 5 official builds

dangelomda, 2025-06-22:
## 🧩 Feature Request: Enable KVM support in Raspberry Pi 5 official builds

### Summary

The Raspberry Pi 5 is a major advancement in performance, boasting a 2.4GHz quad-core Cortex-A76 CPU and up to 16GB RAM. However, it currently lacks native KVM virtualization support in the official Raspberry Pi OS builds. Despite the hardware being fully capable, `/dev/kvm` is not available out-of-the-box.

### Why This Matters

- KVM (Kernel-based Virtual Machine) is a widely-used and mature Linux feature.
- Many users rely on lightweight virtualization for development, security, or edge computing.
- Compiling the kernel manually is a barrier for most users and introduces maintenance challenges.
- Alternative distros may offer support, but lack the stability and community ecosystem of Raspberry Pi OS.

### What We Are Requesting

We respectfully ask that the Raspberry Pi team:

1. Enable KVM support in the default kernel config.
2. Expose `/dev/kvm` in the official Raspberry Pi OS builds.
3. Optionally provide a developer-focused image with virtualization features pre-enabled.
4. Offer basic documentation to help users get started with KVM on Raspberry Pi 5.

### Benefits

- Unlocks full hardware potential of the Pi 5.
- Enables support for common developer workflows (e.g., Home Assistant OS, FreePBX, etc.).
- Aligns Raspberry Pi OS with expectations from a 2025-level development platform.

### Final Note

The Pi 5 deserves official KVM support. Many users are eager to utilize it as a true edge computing platform and not just as a traditional SBC. We are grateful for the incredible work already done and look forward to this being considered.

Thank you!

XECDesign, 2025-06-23:
It is enabled. What's the output of `uname -a`?

https://github.com/raspberrypi/linux/blob/304e1d5171065f013845fc3b8d2866f4bf697f59/arch/arm64/configs/bcm2712_defconfig#L76

```
 ls -la /dev/kvm 
crw-rw----+ 1 root kvm 10, 232 Jun 18 14:14 /dev/kvm
```

dangelomda, 2025-06-23:
**Thank you for the quick reply!**

I confirm that `/dev/kvm` is present on my system and the kernel version is:

```
Linux mypc 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
```

However, when I try to load the `kvm_arm64` module manually:

```
sudo modprobe kvm_arm64
```

I get the following error:

```
modprobe: FATAL: Module kvm_arm64 not found in directory /lib/modules/6.12.25+rpt-rpi-2712
```

This suggests that although the `/dev/kvm` interface is created, the actual kernel module was not included in this build.

Are there any plans to enable `CONFIG_KVM_ARM64` in the official 64-bit Raspberry Pi OS builds?

This would greatly benefit developers who want to unlock the full edge computing potential of the Pi 5.

Thanks again for your work and support!

XECDesign, 2025-06-23:
Where are you seeing references to a module with that name or that config option? `CONFIG_KVM_ARM64` doesn't exist.

closed 2025-06-24 by XECDesign